RoboSense-LiDAR/ros_rslidar #3: NAN classification for data too close or too far

bochen87, 2018-06-13:
In your pointcloud converter, you assign NAN value for point too close or too far away: https://github.com/RoboSense-LiDAR/ros_rslidar/blob/df3332b0c77bec58197cafc231b5f735c5276ffc/rslidar_pointcloud/src/rawdata.cc#L418

However, according to [ROS REP 117](http://www.ros.org/reps/rep-0117.html) the following logic should be implemented:

> Detections that are too close to the sensor to quantify shall be represented by -Inf. Erroneous detections shall be represented by quiet (non-signaling) NaNs. Finally, out of range detections will be represented by +Inf.

Tony-HIT, 2018-06-20:
For the too close point we can know the point is too close. But for the erroneous and too far point we can not know what kind of the point it is at current, so we use the NAN as the invalid points for all.

bochen87, 2018-06-20:
Can you not change the logic that 
`if (dis < DISTANCE_MIN) -> -Inf` and `if (dis > DISTANCE_MIN) -> +Inf or NAN`?

Tony-HIT, 2018-06-21:
But between +Inf and NAN, we can not distinguish them, because they both output a very great value.

bochen87, 2018-06-21:
Ok, understood. How about for that you just put an option to either output +Inf or NAN and write in the readme instructions, so the users can then decide by themselves how to deal with it?

closed 2019-04-24 by Tony-HIT