RoboSense-LiDAR/ros_rslidar #4: Wrong build dependencies

bochen87, 2018-06-13:
In rslidar_driver, you've got a catkin build dependency on rslidar_input https://github.com/RoboSense-LiDAR/ros_rslidar/blob/df3332b0c77bec58197cafc231b5f735c5276ffc/rslidar_driver/package.xml#L29
This catkin package doesn't exist

When running `rosdep install --from-paths /path/to/ws/src --ignore-src` this creates an error 
```
rslidar_driver: Cannot locate rosdep definition for [rslidar_input] 
```

Tony-HIT, 2018-06-20:
we can delete this sentence in the package.xml

bochen87, 2018-06-20:
please delete

Tony-HIT, 2018-06-21:
Please check the latest commit, already delete.

bochen87, 2018-06-21:
Great, thanks!

closed 2018-06-21 by bochen87

james-ward, 2019-05-06:
This issue was reintroduced in commit:
d93f1d7493b8819b2f158241a2cc05e020bde451

facontidavide, 2019-09-25:
Indeed https://github.com/RoboSense-LiDAR/ros_rslidar/blob/master/rslidar_driver/package.xml#L29
:(