RoboSense-LiDAR/ros_rslidar #11: Changing horizontal FOV

pedghz, 2018-12-07:
Hi,

Is it possible to change the horizontal FOV of the RS-LiDAR-16? I want to get 90 degrees of FOV, if it is possible can you help me with the configuration?

Best,
Pedram

closed 2019-04-24 by Tony-HIT

saikishor, 2019-10-30:
@Tony-HIT is this issue fixed?

saikishor, 2019-10-30:
@Tony-HIT I don't think your fixes has really fixed this issue.

ok-kewei, 2023-03-27:
The field of view (FOV) of the Robosense RS-Lidar 16 is determined by the physical design of the sensor, and it cannot be easily modified by the user. However, there are a few things you can do to adjust the FOV of the lidar.

Change the mounting position: By changing the position of the lidar sensor, you can adjust the FOV to some extent. For example, if you mount the sensor higher, the FOV will cover a larger area, while mounting it lower will result in a narrower FOV.

Use software to adjust the data output: The Robosense RS-Lidar 16 outputs data in a specific format, and you can use software to adjust the data to match your desired FOV. For example, you can filter out points that fall outside of your desired FOV or adjust the range limits of the sensor to match the FOV you need.

Use multiple sensors: If you need a wider FOV than what a single Robosense RS-Lidar 16 can provide, you can use multiple sensors and combine their data to create a wider FOV. This approach is commonly used in applications like autonomous driving, where a 360-degree view of the environment is required.

It's important to note that modifying the FOV of a lidar sensor can have implications for its accuracy and performance, so it's recommended to consult with the manufacturer or an expert before making any changes.