RoboSense-LiDAR/ros_rslidar #13: why intensity value is alway zero

planckwu, 2019-01-17:
I got a robosense 16-e lidar, but found that the intensity value is alway zero in the pointcloud2 outputed by rslidar_pointcloud

bryanibit, 2019-03-13:
Because the function has uncorrect transformation of intensity in file **rawdata.cc**:
```
float RawData::calibrateIntensity(float intensity, int calIdx, int distance) 
```
The above function will return zero always whatever the intensity is!

Tony-HIT, 2019-04-24:
It seems you use the wrong calibration file or intensity mode. Have you already solved the problem from robosense technical support.

bryanibit, 2019-04-25:
I *solved* it via **uncommenting** calibrateIntensity function. I don't think I used wrong calibration file. And how and where to change intensity mode?