RoboSense-LiDAR/ros_rslidar #14: Whats the Equivalent Node of transform_node (velodyne) in rslidar?

GeoBIMpro, 2019-02-18:
Hi,
I have bag files with rslidar packets.
I want to convert them to sensor_msgs/PointCloud2 format. Do you have a sample launch file that?

HaoQChen, 2019-08-26:
> Hi,
> I have bag files with rslidar packets.
> I want to convert them to sensor_msgs/PointCloud2 format. Do you have a sample launch file to do that?
> Also, Resulting bag files are larger than Velodyne, may be due to Longer range supported by RSlidar. Is there a ros param to define Maximum range?

Maybe you can record data as a pcap using wireshark, and use the driver by setting "pcap" parameter. Or record rosbag with `rslidarScan` message and use only `rslidar_node`.

donrv, 2020-08-01:
@HaoQChen , I have used the pcap parameter but points are being read slowly. Driver is publishing 75 packets per scan. Is there any setting to read data at actual speed?

HaoQChen, 2020-08-03:
> @HaoQChen , I have used the pcap parameter but points are being read slowly. Driver is publishing 75 packets per scan. Is there any setting to read data at actual speed? Also, is there any way to preserve original time stamps?

75 per scan is right if it is RS-Lidar-16. You can read about `（MSOP）` in the user's manual. There is a param `read_fast`, which mean read data as fast ad possible without sleep.

If you need to preserve original time stamps, I recommend you record `/rslidar_packets` that launch as this:

```
<launch>
  <arg name="model" default="RS16" />
  <arg name="lidar_param_path" default="$(find rslidar_pointcloud)/data/rs_lidar_16/"/>

  <node  name="cloud_node" pkg="rslidar_pointcloud" type="cloud_node" output="screen" >
    <param name="model" value="$(arg model)"/>
    <param name="curves_path" value="$(arg lidar_param_path)/curves.csv" />
    <param name="angle_path" value="$(arg lidar_param_path)/angle.csv" />
    <param name="channel_path" value="$(arg lidar_param_path)/ChannelNum.csv" />
    <param name="max_distance" value="200"/>
    <param name="min_distance" value="0.4"/>
    <param name="resolution_type" value="0.5cm"/>
    <param name="intensity_mode" value="1"/>
  </node>

  <node name="rviz" pkg="rviz" type="rviz"  args="-d $(find rslidar_pointcloud)/rviz_cfg/rslidar.rviz" />
  <node name="rosbag_play" pkg="rosbag" type="play" args="-l YOUR_PATH_TO_BAG"/>

</launch>

```