RoboSense-LiDAR/ros_rslidar #27: Adding Ring information

facontidavide, 2019-07-28:
Hi,

I think it would be nice to add the ring information to the pointcloud, similarly to Velodyne and Ouster.

The changes are pretty simple and I can easily send a PR myself, if you want.

What do you think?

JangGyoungHoon, 2019-08-01:
@facontidavide 

hi, Do you have a ring information added ros driver code.

If you can possible, could you give me the your code. 

my mail address is kert88888888@gmail.com

Thanks,

gyounghoon.

facontidavide, 2019-08-01:
I would rather create a branch here https://github.com/facontidavide/ros_rslidar

I will let you know

facontidavide, 2019-08-01:
I add NO opportunity yet to test this (because I don't have the physical sensor with me this week), but here you can find the preliminary code.

https://github.com/facontidavide/ros_rslidar/tree/ring

@JangGyoungHoon  it would be nice if you test it with RViz and see if the ring field works es expected. Let me know.

facontidavide, 2019-08-14:
For your information, the update code is here https://github.com/facontidavide/ros_rslidar/tree/add_ring_to_pointcloud

TillBeemelmanns, 2020-11-03:
Hi @facontidavide ,
thanks for providing the preliminary code for the ring information. Unfortunately, when I try your code, I do not obtain correct results using the RS32. The ring numbers are not in order.
[image]

Any chance that you could revisit your code ?

facontidavide, 2020-11-03:
I moved to another job and unfortunately I don't have anymore any physical access to one of your sensors. I am afraid I don't have the ability to properly test this anymore

nubertj, 2022-09-29:
Is there any update on this?
Having the ring information would be super useful, in particular if rings are not uniformly distributed.