RoboSense-LiDAR/ros_rslidar #28: Why do we add NAN points instead of just skipping them?

facontidavide, 2019-08-02:
I am referring to this part of the code.... https://github.com/RoboSense-LiDAR/ros_rslidar/blob/master/rslidar_pointcloud/src/rawdata.cc#L873-L882

Wouldn't be easier and more efficient to just skip the point ad, for instance, the Velodyne drive does?

This might be related to issue #27, since people might rely on  the index of the point to determine the ring and removing the NAN points might be an issue for them.

YoshuaNava, 2020-03-06:
Hi @facontidavide 

I think the Velodyne guys were actually going in this direction. Adding NaN to non-existent points in a point cloud is a way of "organizing the point cloud", such that all computations you do can freely be done row and column-wise, and the size of each dataframe can be computed in advance.

facontidavide, 2020-03-06:
ok. I guess it might be an acceptable tradeoff...

closed 2020-03-06 by facontidavide