RoboSense-LiDAR/ros_rslidar #30: ROS Message per point instead of one revolution

donrv, 2019-08-17:
Hi,
Is there any setting to output ROS pointcloud2 message per point instead of per revolution?
Currently, my RS16 is set to 600 RPM and getting 10 Messages per second. I want to read messages in realtime instead of accumulating per revolution.

HaoQChen, 2019-08-26:
> Hi,
> Is there any setting to output ROS pointcloud2 message per point instead of per revolution?
> Currently, my RS16 is set to 600 RPM and getting 10 Messages per second. I want to read messages in realtime instead of accumulating per revolution.

maybe you can change `config_.npackets` in `rslidarDriver::poll` from rsdriver.cpp. This will change the package number of each scan

donrv, 2019-09-02:
@HaoQChen Thank you, Got it

closed 2020-02-11 by donrv