RoboSense-LiDAR/ros_rslidar #31: [driver][socket] Rslidar poll() timeout Ubuntu 18.04 ROS Melodic

mfogelson, 2019-09-06:
I know that you only explicitly support ubuntu 14.04 and 16.04, but I have been trying to set up the rs_lidar_16 on 18.04 but get this warning when running :

```
roslaunch rslidar_pointcloud rs_lidar_16.launch
```

> [ INFO] [1567733053.266502478]: [driver] publishing 75 packets per scan
[ INFO] [1567733053.268930206]: [driver] Cut at specific angle feature activated. Cutting rslidar points always at 0 degree.
process[rviz-4]: started with pid [12122]
[ INFO] [1567733053.275715743]: [driver][input] accepting packets from IP address: 192.168.1.200
[ INFO] [1567733053.275763845]: [driver][socket] Opening UDP socket: port 6699
[ INFO] [1567733053.277135546]: [driver][input] accepting packets from IP address: 192.168.1.200
[ INFO] [1567733053.277165407]: [driver][socket] Opening UDP socket: port 7788
[ INFO] [1567733053.281332242]: [cloud][rawdata] lidar model: RS16
[ INFO] [1567733053.281799509]: [cloud][rawdata] start and end angle feature activated.
[ INFO] [1567733053.281820653]: [cloud][rawdata] start_angle: 0 end_angle: 360 angle_flag: 1
[ INFO] [1567733053.282580738]: [cloud][rawdata] distance threshlod, max: 200 m, min: 0.4 m
[ INFO] [1567733053.283324075]: [cloud][rawdata] initialize resolution type: 0.5 cm, intensity mode: 1
[ WARN] [1567733054.280001802]: [driver][socket] Rslidar poll() timeout

Is there any solutions for this issue?

GeneraalAlfa, 2020-03-12:
I had the some problem. You need to check if you can ping your RS-LiDAR (IP: 192.168.1.200). It can be that you have not you Ethernet port to the specified IP-address. Change the IP-address of the Ethernet port you have your sensor connected on to the static IP 192.168.1.102.

After that check that it is the above IP-address and try to ping you RS-LiDAR again. If you get a result/ping back you can try the roslaunch again.

I hope my explication is understandable and good luck.

Qin-xs, 2020-05-29:
I also encountered this problem. I set the IP correctly and can ping the device, but the following error still occurs.

[ INFO] [1590716480.064869319]: [driver] publishing 75 packets per scan
process[rviz-4]: started with pid [16454]
[ INFO] [1590716480.107415083]: [driver] Cut at specific angle feature activated. Cutting rslidar points always at 0 degree.
[ INFO] [1590716480.121549441]: [cloud][rawdata] lidar model: RS16
[ INFO] [1590716480.125510217]: [cloud][rawdata] start and end angle feature activated.
[ INFO] [1590716480.125638317]: [cloud][rawdata] start_angle: 0 end_angle: 360 angle_flag: 1
[ INFO] [1590716480.132764329]: [cloud][rawdata] distance threshlod, max: 200 m, min: 0.4 m
[ INFO] [1590716480.140216558]: [driver][input] accepting packets from IP address: 192.168.1.200
[ INFO] [1590716480.140435923]: [driver][socket] Opening UDP socket: port 6699
[ INFO] [1590716480.141402797]: [cloud][rawdata] initialize resolution type: 0.5 cm, intensity mode: 1
[ INFO] [1590716480.143887823]: [driver][input] accepting packets from IP address: 192.168.1.200
[ INFO] [1590716480.144069107]: [driver][socket] Opening UDP socket: port 7788
[ INFO] [1590716480.324866523]: [cloud][rawdata] angle data is wrote in difop packet!
[ INFO] [1590716480.325391017]: [cloud][rawdata] distance resolution is: 0.5 cm, intensity mode is: Mode 3
[ INFO] [1590716480.325687665]: [cloud][rawdata] lidar support dual return wave, the current mode is: strongest
[ INFO] [1590716480.325795124]: [cloud][rawdata] difop intensity mode: 3
Error:   Received an invalid TCPROS header.  Each line must have an equals sign.
         at line 103 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.13/src/header.cpp

TobiMiller, 2020-07-02:
anyone got this error fixed? we can ping the sensor but still timeouts

DeepDuke, 2020-07-25:
I met the same problem on melodic too. I set the correct  ip, but i couldn't ping the lidar.

HaoQChen, 2020-08-03:
> I met the same problem on melodic too. I set the correct ip, but i couldn't ping the lidar.

Did you change the MSOP port or DIFOP port? You can run a Wireshark, and check if your setting is right. If you are new to this driver, you can use our new driver, which support melodic and ROS2 formally. 

https://github.com/RoboSense-LiDAR/rslidar_sdk

czp0204, 2024-05-24:
> I met the same problem on melodic too. I set the correct ip, but i couldn't ping the lidar.

Dude, did u fixed that problem? How to set IP on this device? the launch file ?