RoboSense-LiDAR/ros_rslidar #32: How to restrict the FOV from the launch file

saikishor, 2019-10-30:
Can someone tell me how to restrict the FOV of the sensor from the launch file. I see a similar question posted here https://github.com/RoboSense-LiDAR/ros_rslidar/issues/11, but the issue is closed with some fixes, I am not sure if the issue is really fixed. Can someone post their insight on this issue?.

Thanks in advance.

HaoQChen, 2019-12-25:
You can set it with `start_angle` and `end_angle` param. They are read in here https://github.com/RoboSense-LiDAR/ros_rslidar/blob/b70f7a6a013ca37ecf4a3268ec8e8e29e7a56f45/rslidar_pointcloud/src/rawdata.cc#L42

and limit horizontal FOV in here https://github.com/RoboSense-LiDAR/ros_rslidar/blob/b70f7a6a013ca37ecf4a3268ec8e8e29e7a56f45/rslidar_pointcloud/src/rawdata.cc#L936

This is an example to limit the horiz angle from 320° to 40°:

```html
  <node  name="cloud_node" pkg="rslidar_pointcloud" type="cloud_node" output="screen" >
    <param name="model" value="$(arg model)"/>
    <param name="curves_path" value="$(arg lidar_param_path)/curves.csv" />
    <param name="angle_path" value="$(arg lidar_param_path)/angle.csv" />
    <param name="channel_path" value="$(arg lidar_param_path)/ChannelNum.csv" />
    <param name="max_distance" value="200"/>
    <param name="min_distance" value="0.4"/>
    <param name="resolution_type" value="0.5cm"/>
    <param name="intensity_mode" value="1"/>
    <param name="start_angle" value="320"/>
     <param name="end_angle" value="40"/>
  </node>
```

saikishor, 2019-12-25:
@HaoQChen Thanks for fix, it should be really helpful. I will give it a try at the earliest and get back to you.

saikishor, 2020-01-16:
@HaoQChen Sorry for the delay, It worked. Thank you.

closed 2020-01-16 by saikishor