RoboSense-LiDAR/ros_rslidar #35: Launching two RS16 together gives same frame I'd "rslidar"

donrv, 2019-12-21:
Hi,
When Launching two RS16 lidars together, ns1 and ns2 topics have same frame I'd "rslidar". Can we set different frame id's in the launch file itself?

HaoQChen, 2019-12-25:
Can I have a look at your launch file?

HaoQChen, 2019-12-25:
You can set `tf_prefix` param according to this 

https://github.com/RoboSense-LiDAR/ros_rslidar/blob/b70f7a6a013ca37ecf4a3268ec8e8e29e7a56f45/rslidar_driver/src/rsdriver.cpp#L29

Here is an example:
```html
  <node  name="rslidar_node" pkg="rslidar_driver" type="rslidar_node" output="screen" >
    <param name="model" value="$(arg model)"/>
    <param name="device_ip" value="$(arg device_ip)" />
    <param name="msop_port" value="$(arg msop_port)" />
    <param name="difop_port" value="$(arg difop_port)"/>
    <param name="cut_angle" value="$(arg cut_angle)"/>
    <!--param name="pcap" value="path_to_pcap"/-->
    <param name="tf_prefix" value="ns1"/>
  </node>
```

lmunier, 2020-01-27:
Hello,

A fix that works on ros melodic is to set the "frame_id" parameter in your launch file. It will not change your topic names but will let you put tf with the news "frame_id" as references.

Best regards,
Louis

donrv, 2020-02-11:
Thanks @HaoQChen

closed 2020-02-11 by donrv