RoboSense-LiDAR/ros_rslidar #39: Fixed frame does not exist

qrort, 2020-05-11:
I have successfully downloaded and built the driver. I am trying to visualise pointcloud from `.pcap `file according to the tutorial. I have edited the `.launch` file and typed in the cmd `roslaunch rslidar_pointcloud rs_lidar_32.launch`. There are no points on rviz grid as there is the error
`No tf data. Actual error: Fixed Frame [rslidar] does not exist`. I am really new to ROS and I am struggling to fix that

sergorl, 2020-06-21:
@qrort , did you fix it?

qrort, 2020-06-28:
Unfortunately, no; it is no more relevant though. I could guess it happened because I was launching ROS on a virtual machine, but I had no other choice at the moment.

closed 2020-06-28 by qrort

ok-kewei, 2023-05-02:
@sergorl , do you have the answer for that ?

ok-kewei, 2023-05-02:
Under RVIZ > Displays> Global Options > Fixed Frame, change to 'rslidar'.