RoboSense-LiDAR/ros_rslidar #40: May I ask for help about how to understand different intensity mode?

FishInWave, 2020-05-14:
I have read the code below in raw_data.cc:
`if ((data[291] == 0x00) || (data[291] == 0xff) || (data[291] == 0xa1))
    {
      intensity_mode_ = 1;  // mode for the top firmware lower than T6R23V8(16) or T9R23V6(32)
    }
    else if (data[291] == 0xb1)
    {
      intensity_mode_ = 2;  // mode for the top firmware higher than T6R23V8(16) or T9R23V6(32)
    }
    else if (data[291] == 0xc1)
    {
      intensity_mode_ = 3;  // mode for the top firmware higher than T6R23V9
    }`
Do T6R23V8(16) and T9R23V6(32) represent the version of firmware?
If so, I have no idea aout what version of my RS-LiDAR 16. The manufacture date of my Lidar is 21/9/2019.
Hoping your reply, Thanks.

HaoQChen, 2020-08-03:
> I have read the code below in raw_data.cc:
> `if ((data[291] == 0x00) || (data[291] == 0xff) || (data[291] == 0xa1)) { intensity_mode_ = 1; // mode for the top firmware lower than T6R23V8(16) or [...]

Yes, it is the version of firmware. `data` is DIFOP DATA, lidar will send it to driver. It just tells us which intensity mode to use and how to get it. It base on its firmware

Oyssster, 2021-09-25:
> Yes, it is the version of firmware. `data` is DIFOP DATA, lidar will send it to driver. It just tells us which intensity mode to use and how to get it. It base on its firmware
Dose it mean the intensity_mode_ can't be set by the users?