RoboSense-LiDAR/ros_rslidar #42: no tf data recieved

sergorl, 2020-06-20:
I did
`rosbag play -l some.bag`
and
`rostopic echo /left/rslidar_packets` showed some data.

But when I launched `rviz`, it showed `NO TF Data`.

Also `rosrun tf view_frames` gives: `no tf data recieved`

I use ROS-noetic and ubuntu 20.04.

How can I fix it?

HaoQChen, 2020-08-03:
We have not tested code on Noetic yet~~

Did you change fixed frame to `rslidar` like this?

[image]

Prvn1201, 2022-07-20:
i have the same problem. I have changed the fixed frame to rslidar but still the same.

ok-kewei, 2023-05-02:
any solution on that ?

ok-kewei, 2023-05-02:
does anyone have the answer for that ?