RoboSense-LiDAR/ros_rslidar #44: Rslidar poll() timeout for two_lidar.launch example

HTLife, 2020-07-01:
I just met the problem that when I try to use the sample launch file to get data from two lidar.

I'm sure that each lidar is setting correctly because when I comment out one group in the launch file, another lidar just works fine.
The problem is I couldn't enable two group in the same time.

Any possible cause of this problem?

JinxiongLu, 2020-07-20:
I encounter the same problem at the moment, craving for a solution

HaoQChen, 2020-08-03:
Did you fix this?

It does not make sense. Did you test both lidars separately?

HTLife, 2020-08-06:
I resolved this problem.

The problem is I need to set different port for 
```
<param name="msop_port" value="9966"/>
    <param name="difop_port" value="8877"/>
```
two lidar.

Does this make sense?  even they have two different IP?

closed 2020-08-06 by HTLife