RoboSense-LiDAR/ros_rslidar #47: Is it compatible with rs-lidar-M1?

RomanRobotnik, 2020-09-08:
Hi,

Is it working for [this model](https://www.robosense.ai/rslidar/rs-lidar-M1)?

Maybe we just need to add a new rslidar_pointcloud configuration like [these ones](https://github.com/RoboSense-LiDAR/ros_rslidar/tree/master/rslidar_pointcloud/data)...

Thank you!

HaoQChen, 2020-09-24:
We developed a new driver which is going to be compatible with M1 in the future.

https://github.com/RoboSense-LiDAR/rs_driver

We plan to  release in October

je5349, 2021-01-07:
> We developed a new driver which is going to be compatible with M1 in the future.
> 
> https://github.com/RoboSense-LiDAR/rs_driver
> 
> We plan to release in October

@HaoQChen 

hello, i also have same question like questioner.
I'm beginner in this field. So i don't know well how to use rs_driver.
And i want to use RS-LiDAR-M1 with ROS.
Can you tell me how to do?

HaoQChen, 2021-01-07:
> @HaoQChen
>
> hello, i also have same question like questioner.
> I'm beginner in this field. So i don't know well how to use rs_driver.
> And i want to use RS-LiDAR-M1 with ROS.
> Can you tell me how to do?

https://github.com/RoboSense-LiDAR/rslidar_sdk  this project will help you to connect our lidar and public ROS message. You can try with the README and guide. 

Please contact our technical support, sales or issue on github if you feel any confuse.