RoboSense-LiDAR/ros_rslidar #53: [driver][socket] Rslidar poll() timeout

aprilliuwei, 2021-11-13:
hi,i got a prrblem when i use the rs_lidar_16 like this,can u give me some advices
!thanks~[Screenshot from 2021-11-13 19-34-38](https://user-images.githubusercontent.com/55389624/141642349-554f8aec-a636-4f06-950d-dc27dcc3d4e5.png)
[image]

HomieRegina, 2022-07-06:
Hi, I get the same issue now.  Do you solve it ?

waa-7913, 2024-11-07:
我遇到过相同的问题Rslidar poll() timeout，主要是将型号搞混了，RS16和RS-helios-16p，建议先确定自己的激光雷达型号，在激光雷达的底部，之后配置好ip地址和端口后使用官方提供的rsview.exe，可以尝试看是否有点云，成功之后再使用ros包尝试。
[image]